Makeblock-official/mDrawBot #11: Mdraw OSX Arm Moving wen pointed. But no draw svg

olivierlewaiss, 2015-05-26:
Hi,

I changed the USB (the one in the box wasn't working). The arm is moving when I point a place on mDraw.

The servo pen goes only up when I enter new settings, but not down.

When I import a svg file. Nothing happens. I click on the play button, which becomes green on click, then blue again, but nothing happens.

software version is version=1.06 2015-5-08. The port I connect on is /dev/cu.wchusbserial1420, I don’t find COM20.

could you please tell me what’s wrong, someone using Mdraw on a Mac ?

Thank you
Oivier

Makeblock-official, 2015-05-27:
there is some svg files for test purpose here:
https://github.com/Makeblock-official/mDrawBot/tree/master/mDrawGui/svg%20test
please let me know if it works for you

olivierlewaiss, 2015-05-27:
Hi,

could you tell me how to download an svg file from the github ? I only access to the code. There is no downloas button on github…Anyway I will try with another SVG...

Thanks
--
Olivier Waissmann

> Le 27 mai 2015 à 03:56, Makeblock <notifications@github.com> a écrit :
> 
> https://github.com/Makeblock-official/mDrawBot/tree/master/mDrawGui/svg%20test

olivierlewaiss, 2015-05-27:
Hi again,

I downloaded a SVG file from Internet and it is finally working. I think there is something wrong with SVG files generated by Illustrator Mac but I don’t know wat…

How can I buy the bluetooth module ?

Thank you

regards
--
Olivier Waissmann