Makeblock-official/mDrawBot #6: mScara arm tries to go 360 degrees

yzxj, 2015-05-26:
The construction of the base restricts the arm to a movement range of about -90deg to 160deg. This restricts the drawing space to mostly the left side of the robot, and anything on the right ends up causing the arm to go anticlockwise until it hits the base and gets stuck.

schmurfy, 2015-05-26:
I also had some issues with it, It would be nice if we could tell the IDE somehow the physical constraints so it won't try to move past them.
I think there is any easy way: ask the users to position the arm in the middle (perpendicular to the base of the table where its base sits) and then deduce the useable arc, that way you don't need complex calibration step(s).

yzxj, 2015-05-26:
Positioning the arm in the middle may help, but it doesn't solve the underlying problem: the right half of the drawing area is virtually inaccessible because (1) the further segment of the arm always bends leftwards, and (2) the arm gets blocked by the base when attempting to reach something close to the base but on the right. For example, a point 10 cm from the base and at 135 degrees is inaccessible even if the arm does not go around the back.
[image]

If it were to bend the other direction, it would have access to the right half without needing to go 360. I'm not sure about how much calibration and rewriting of the code this would require, though. An alternative would be to switch the arm axis to the right side of the base, which, although less ideal (the 'home' position cannot ever be at the 0 degree position), would require fewer changes to the code.

schmurfy, 2015-05-26:
I understand your problem but I feel there are two different problems:
1) The software does not know how much the arm can turn safely
2) The software cannot command the arms to reach certain locations with the mechanical constraints, maybe the arm could bend one way or the other depending on which quarter of the circle the target point is in but it might become really complex if you draw between points in different quarters xD

[image]

As a side note I feel like the hardware design could be adjusted to give more freedom to the arm but it might require more pieces than the kit has.

Ps: nice schema :)

Cration, 2015-05-27:
I agree with @schmurfy . If you change the structure of mScara a little, the software will never know the safety zone to move.

asianboy75, 2015-05-28:
I've got the same problem.

schmurfy, 2015-05-31:
The IDE should at least prevent doing impossible movemenst, just limit the space where we can click or place images, that's an easy and would prevent this issue from happenning.