Makeblock-official/mDrawBot #7: Installing mDraw on Linux/Ubuntu

Buitani, 2015-05-26:
How to install the mDraw Software on Ubuntu?

Buitani, 2015-05-26:
What I did so far.
1. Downloaded the mDraw Zip from Github

2. Installed Python
> sudo apt-get install python 

3. Went to the mDrawGui Directory

4. Didnt know which file to start with python. But it looks like I found the right one. This works for me:
> python robot_gui.py

Makeblock-official, 2015-05-27:
yes, maybe you have to run it from source code for now, and update firmware by arduino ide

Buitani, 2015-05-27:
How do I do that? How do I run the software from source code?

np, 2015-05-27:
@Buitani  This is what you just did.
@Makeblock-official any starting point you would recommend about linux and arduino?

seken, 2015-06-08:
I have been working on this and have a way of running the source code, but I'm still working out all the issues on Linux. Firmware upload works, but connection is sporadic due to the line endings. I've written a test to prove why the connection sometimes worked, just need to update the original code

BugBuggi, 2015-06-15:
When will a Linux Package be available? I have the Kit since a week now, robot is built but of no use as I only run Ubuntu on my Computers...
Open Source Software should always be available for all three Platforms, Linux at first, as it is the only Open source operating System of the three!

I will be glad to test if you can provide any package!

seken, 2015-06-15:
I've added the necessary code but its not currently built into a package, the svg opening issue is still there, but it runs. I've only tested the eggbot, but I don't see what the difference would be in the other modes.

meeuw, 2015-06-17:
To open SVG's in Linux replace the following line in robot_gui.py
from:
            filename = str(QtGui.QFileDialog.getOpenFileName(self, 'Open Svg/Bmp', '', ".svg;.bmp(*.svg;*.bmp)").toUtf8())
to:
            filename = str(QtGui.QFileDialog.getOpenFileName(self, 'Open Svg/Bmp', '', ".svg(*.svg)").toUtf8())

Buitani, 2015-09-04:
Hi there,

since the software got updated there is no robot_gui.py in the mDrawGui Folder.

Which ist the correct file to run mDraw?

I did run RobotUtils.py with

> python RobotUtils.py 

Nothing happend.

I did run mDraw.Gui with

> python  mDraw.Gui

and got following message:

> Traceback (most recent call last):
>   File "mDraw.py", line 5, in <module>
>    import SvgParser
>   File " /mdrawbot/mDrawGui/SvgParser.py", line 5, in <module>
>     from PyQt5.QtGui import*
> ImportError: No module named PyQt5.QtGui

meeuw, 2015-09-05:
I'm using python3 mDrawGui/mDraw.py to start mDraw.

You also need to install python-qt5.

Buitani, 2015-09-05:
Hi meeuw,

thank you for your reply. Now I installed python-qt5 and tried to start mDraw with

> python3 mDraw.py

I got following message:

>Traceback (most recent call last):
>  File "mDraw.py", line 4, in <module>
>    import SerialCom
>  /mdrawbot/mDrawGui/SerialCom.py", line 4, in <module>
>   import serial
> ImportError: No module named 'serial'

meeuw, 2015-09-05:
You might have guessed it, you should install pyserial for the serial module.

Buitani, 2015-09-05:
You were right. I had modules for python installed but not for phyton3. Now I installed the missing modules and mDraw is working. (Sometimes it crashes, but mostly its working) 

Opening files looks a little bit strange, because background is in dark grey and the fonts are black, so its hard to read.

Also I still have that problem with the laser which I also reported here [#22](https://github.com/Makeblock-official/mDrawBot/issues/22). mSacra moves ridiculous fast when changing delay. (Even faster like fastest speed possible in the speed setup) I dont know if its a problem with mDraw in general or just with linux/ubuntu. Result: I cant do anything productive with the laser.

alexanderadam, 2017-04-16:
I am [unable to get it working](http://forum.makeblock.com/t/no-communication-between-computer-and-mscara/3048/5?u=noconnection) but at some point I got this in console output:

```
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/home/user/mDrawBot/mDrawGui/SerialCom.py", line 49, in run
    l = self.ser.readline().decode('utf-8')
  File "/home/user/.local/lib/python3.5/site-packages/serial/serialposix.py", line 493, in read
    buf = os.read(self.fd, size - len(read))
TypeError: an integer is required (got type NoneType)
```

Does anybody has any idea whether these issue could be related?

meeuw, 2017-04-16:
Hi @alexanderadam I'm guessing here but do you have the right permission to your serial port? Which device in /dev are you using and what are the permissions for this device file?

alexanderadam, 2017-04-16:
I'm starting it with sudo rights at the moment… 😒

giancarlocp, 2019-01-06:
Package needed:
$ apt install python3-pyqt5 python3-pyqt5.qtsvg python3-serial
$ cd mdraw; python3 mDraw.py

matthieu-labas, 2019-01-14:
> Hi @alexanderadam I'm guessing here but do you have the right permission to your serial port? Which device in /dev are you using and what are the permissions for this device file?

You need to be in the `dialout` group:

    sudo usermod -G dialout <login>

matthieu-labas, 2019-01-14:
> Package needed:
> $ sudo apt install python3-pyqt5 python3-pyqt5.qtsvg python3-serial
> $ python3 mDrawGui/mDraw.py

And package `python3-tk` in case it was not installed...

alexanderadam, 2019-01-14:
@matthieu-labas I can't check anymore. I wrote about this issue nearly two years ago and I'm not actively using it anymore at the moment. I also upgraded my installation in the meantime so it could be that things changed anyway. :wink: