Makeblock-official/mDrawBot #8: Using OS-X to connect with USB not working?

palerikm, 2015-05-26:
Running 14.3.0 Darwin Kernel Version 14.3.0: Mon Mar 23 11:59:05 PDT 2015; root:xnu-2782.20.48~5/RELEASE_X86_64 x86_64

After installing the USB drivers your documentation points to I do get an additional device I can connect to (/dev/cu.Makeblock-ELETSPP), however that device is always reported as busy when I try to connect.

version=1.06 2015-5-08
[Errno 16] Resource busy: '/dev/cu.Makeblock-ELETSPP'

Please let me know if there is anything I can help test to resolve the issue.

olivierlewaiss, 2015-05-26:
Hi,

I changed the USB (the one in the box wasn't working). The arm is moving when I point a place on mDraw.

The servo pen goes only up when I enter new settings, but not down.

When I import a svg file. Nothing happens. I click on the play button, which becomes green on click, then blue again, but nothing happens.

software version is version=1.06 2015-5-08. The port I connect on is /dev/cu.wchusbserial1420, I don’t find COM20.

could you please tell me what’s wrong, someone using Mdraw on a Mac ?

Thank you
Oivier

Makeblock-official, 2015-05-27:
did the led on bluetooth module stop flashing once you pair or connect to it? for connection problem maybe you can try forum.makeblock.cc

palerikm, 2015-05-27:
This was more related to the USB connection. Have not yet thorogly tested the bluetooth connection from OSX (Briefly tested with an IOS device and that paired up).

Running windows on a virtual machine  on OS-X and then connecting works fine. There are a few hiccups concerning the USB drivers if they are not properly signed. 

The documentation should make it clear that newer versions of OS-X might have problems using the USB driver. (At least it did not work for me as described in the documentation)

Feel free to close the issue. Can take it to the forum and see what happens there.

brainlessboy, 2016-03-13:
i have the same issue, but really? installing windows? next generation kit (this is money on top to by a windows license)
is there a way to get it to work? i definitely ordered this because it said (mac compatible)
help