Makeblock-official/mDrawBot #9: remember position

schmurfy, 2015-05-26:
Each time you connect to the mScara arm the IDE consider the arm is in the Home position, this is really annoying, surely there must be a way when it connect to retrieve the current position from the controller (Arduino/Orion) so that we are not required to return to home every time.

Cration, 2015-05-27:
Each time you connect to mDrawBot, it will restart. There's no position feedback sensors, just two stepper motors on it. So I think it's really improbable to do that. What if you manually move the position of the robot arm?

schmurfy, 2015-05-27:
I was not thinking about any sensor but when your computer is connected to the controller you don't have to return it to home before every command right ? Why the controller could not keep that position in memory to send it back to the computer when connected ?

I am not talking about powering off, just disconnect and reconnect the USB cable and you have to move back the arm to the home position by hand (after removing the power since you can't move them otherwise), really annoying.

By going further I think the position could be stored in the eeprom but that's another story.

waldschip, 2015-05-27:
the problem is that the eeprom have a life cycle, i think it would be better mount angular sensor to the joints

schmurfy, 2015-05-27:
@Cration I misunderstood the part about the restart, you mean the controller restarts when you connect to the computer ? If so why ?
I have not looked at the code too much for now but the controller code looks "dumb" (I don't mean easy but I mean it does not take any action by itself), it takes commands as inputs and acts accordingly, right ?

If that is the case I don't understand why it would need to reboot unless the power is removed or you want to reflash it.

Cration, 2015-05-28:
@schmurfy   Because they connect the DTR pin to the reset pin of MCU with a capacity on PCB. It is the same as Arduino UNO. I think that's for downloading the firmware with Arduino IDE.

schmurfy, 2015-05-28:
ok, I missed that :p
if I prevent the reset on connect will the IDE be able to get back the current position from the controller ?

closed 2022-03-03 by schmurfy