Makeblock-official/mDrawBot #12: mDraw Test Program Crashes on Startup on Mac System

nilocor, 2015-05-27:
I downloaded the latest version of the software which was supposed to fix this issue, but the application still crashes.

Makeblock-official, 2015-05-27:
would you please paste your crashing log here? and extract the zip to an english-only path, thanks.

Iwantdraw, 2015-05-27:
Same to me. Mac OS X 10.8.5.

Iwantdraw, 2015-05-28:
Here my crashing log:
Process:         robot_gui [18476]
Path:            /Developer/mDrawBot-master/distribution/mac/mDraw.app/Contents/MacOS/robot_gui
Identifier:      robot_gui
Version:         0.0.0 (???)
Code Type:       X86-64 (Native)
Parent Process:  robot_gui [18475]
User ID:         502

Date/Time:       2015-05-28 14:47:16.640 +0200
OS Version:      Mac OS X 10.8.5 (12F2501)
Report Version:  10

Interval Since Last Report:          894977 sec
Crashes Since Last Report:           111
Per-App Crashes Since Last Report:   31
Anonymous UUID:                      B4E42143-055C-BE32-A02D-C80AC2401EA3

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: KERN_INVALID_ADDRESS at 0x00000000f3403da0

VM Regions Near 0xf3403da0:
--> 
    __TEXT                 000000010c0be000-000000010c0db000 [  116K] r-x/rwx SM=COW  /Developer/mDrawBot-master/distribution/mac/mDraw.app/Contents/MacOS/robot_gui

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_c.dylib             	0x00007fff89d97670 strlen + 16
1   libsystem_c.dylib             	0x00007fff89e1ea2d __vfprintf + 4704
2   libsystem_c.dylib             	0x00007fff89e1d2ff vfprintf_l + 727
3   robot_gui                     	0x000000010c0d39e2 0x10c0be000 + 88546
4   robot_gui                     	0x000000010c0c0261 0x10c0be000 + 8801
5   robot_gui                     	0x000000010c0bfb9e 0x10c0be000 + 7070
6   robot_gui                     	0x000000010c0d3b78 0x10c0be000 + 88952
7   robot_gui                     	0x000000010c0bf454 0x10c0be000 + 5204

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff8e4276d6 __workq_kernreturn + 10
1   libsystem_c.dylib             	0x00007fff89dabf1c _pthread_workq_return + 25
2   libsystem_c.dylib             	0x00007fff89dabce3 _pthread_wqthread + 412
3   libsystem_c.dylib             	0x00007fff89d96191 start_wqthread + 13

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8e427d16 kevent + 10
1   libdispatch.dylib             	0x00007fff88c31dea _dispatch_mgr_invoke + 883
2   libdispatch.dylib             	0x00007fff88c319ee _dispatch_mgr_thread + 54

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8e4276d6 __workq_kernreturn + 10
1   libsystem_c.dylib             	0x00007fff89dabf1c _pthread_workq_return + 25
2   libsystem_c.dylib             	0x00007fff89dabce3 _pthread_wqthread + 412
3   libsystem_c.dylib             	0x00007fff89d96191 start_wqthread + 13

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x00000000ffffffff  rbx: 0x000000010c0d47e1  rcx: 0x00000000f3403da5  rdx: 0x00000000f3403da5
  rdi: 0x00000000f3403da0  rsi: 0x00007fff53b402d8  rbp: 0x00007fff53b404f0  rsp: 0x00007fff53b400b8
   r8: 0x000000010c103800   r9: 0x00007fff53b40b40  r10: 0x000000010c0e7000  r11: 0x00007fbdf480401a
  r12: 0x0000000000000000  r13: 0x0000000000000000  r14: 0x00000000ffffffff  r15: 0x0000000000000073
  rip: 0x00007fff89d97670  rfl: 0x0000000000010286  cr2: 0x00000000f3403da0
Logical CPU: 3

Binary Images:
       0x10c0be000 -        0x10c0daff7 +robot_gui (0.0.0 - ???) <C371E49E-CC99-3791-8965-4D3EE55AD3BF> /Developer/mDrawBot-master/distribution/mac/mDraw.app/Contents/MacOS/robot_gui
    0x7fff6bcbe000 -     0x7fff6bcf293f  dyld (210.2.3) <A40597AA-5529-3337-8C09-D8A014EB1578> /usr/lib/dyld
    0x7fff81af5000 -     0x7fff81b77fff  com.apple.Heimdal (3.0 - 2.0) <716C152F-D299-3ACB-BCB0-E51F2D604BD1> /System/Library/PrivateFrameworks/Heimdal.framework/Versions/A/Heimdal
    0x7fff81b78000 -     0x7fff81bd2fff  com.apple.print.framework.PrintCore (8.3 - 387.2) <5BA0CBED-4D80-386A-9646-F835C9805B71> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/Frameworks/PrintCore.framework/Versions/A/PrintCore
    0x7fff81c63000 -     0x7fff81cb4ff7  com.apple.SystemConfiguration (1.12.2 - 1.12.2) <A4341BBD-A330-3A57-8891-E9C1A286A72D> /System/Library/Frameworks/SystemConfiguration.framework/Versions/A/SystemConfiguration
    0x7fff81cb5000 -     0x7fff81cb5fff  libkeymgr.dylib (25) <CC9E3394-BE16-397F-926B-E579B60EE429> /usr/lib/system/libkeymgr.dylib
    0x7fff81cec000 -     0x7fff81cecfff  com.apple.Accelerate.vecLib (3.8 - vecLib 3.8) <F565B686-24E2-39F2-ACC3-C5E4084476BE> /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/vecLib
[... 168 lines cut ...]
__UNICODE                           544K
shared memory                       308K
===========                      =======
TOTAL                             246.5M

Model: MacBookPro8,1, BootROM MBP81.0047.B27, 2 processors, Intel Core i5, 2.4 GHz, 8 GB, SMC 1.68f99
Graphics: Intel HD Graphics 3000, Intel HD Graphics 3000, Built-In, 512 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1333 MHz, 0x855D, 0x202020202020202020202020202020202020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1333 MHz, 0x855D, 0x202020202020202020202020202020202020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xD6), Broadcom BCM43xx 1.0 (5.106.98.100.17)
Bluetooth: Version 4.1.7f5 15113, 3 service, 21 devices, 3 incoming serial ports
Network Service: Ethernet 1, Ethernet, en0
Network Service: AirPort, AirPort, en2
Serial ATA Device: Samsung SSD 850 PRO 512GB, 512,11 GB
Serial ATA Device: MATSHITADVD-R   UJ-8A8
USB Device: FaceTime HD Camera (Built-in), apple_vendor_id, 0x8509, 0xfa200000 / 3
USB Device: hub_device, 0x0424  (SMSC), 0x2513, 0xfa100000 / 2
[... cut ...]
USB Device: Apple Internal Keyboard / Trackpad, apple_vendor_id, 0x0253, 0xfa120000 / 5
USB Device: BRCM2070 Hub, 0x0a5c  (Broadcom Corp.), 0x4500, 0xfa110000 / 4
USB Device: Bluetooth USB Host Controller, apple_vendor_id, 0x821a, 0xfa113000 / 7
USB Device: hub_device, 0x0424  (SMSC), 0x2513, 0xfd100000 / 2
USB Device: Keyboard Hub, apple_vendor_id, 0x1006, 0xfd120000 / 4
USB Device: Apple Keyboard, apple_vendor_id, 0x0221, 0xfd122000 / 5
USB Device: IR Receiver, apple_vendor_id, 0x8242, 0xfd110000 / 3

Makeblock-official, 2015-05-29:
The mac distribution is built under my mac which runs yosemite.
I am looking into mac virtualenv, hoop this will fix the compatible problems.

Iwantdraw, 2015-06-02:
Thank you very much for your effort. What is the status?
I'm very excited to run my mDrawBot very soon.