Makeblock-official/mDrawBot #13: Where can we buy the plastic hinge ?

asianboy75, 2015-05-28:
I've looked in the website but I couldn't find it :(
Thanks

schmurfy, 2015-05-31:
I had three of each in my kit, you already had one ?

asianboy75, 2015-06-01:
I broke the three :(

tipex20, 2015-06-05:
So, it's impossible to replace plastic hinge. Is it the same for the Pen Holder ?

schmurfy, 2015-06-07:
I contacted them by email now a week ago as well as kickstarter message and sill waiting for an answer so I think they might be swamped with requests xD
Either that or they took a break.

tipex20, 2015-06-07:
Hi @schmurfy, I hope that their team will be there to help us. They didn't seem to respond from now.

tipex20, 2015-07-08:
One month later, still waiting for an answer

ajxn, 2025-04-18:
Late, but anyway.  They are plastic sheets, so should be easy to make self.