Makeblock-official/mDrawBot #15: What does this mean when opening a BMP file for engraving

robbo1, 2015-05-31:
When opening a BMP file for engraving  the conversion window opens and it has the following statement in that window

"Note that only laser kit support original format, if you are using pen or other servo driven kit, please translate bitmap to svg path."

and there is a button "Convert to Grey" which seems to do nothing.

Please explain

Makeblock-official, 2015-06-02:
You may find the converted svg in the same path of bmp file. Not sure why converted svg file not show up in the canvas after packing, it seems that pyqt is not the best choice..

robbo1, 2015-06-02:
It seems you missed my point

What does the "Convert to Grey" button do???
-----------------------------------------------------------

And I thought the Laser Engraver could do greyscale engraving