Makeblock-official/mDrawBot #18: won't load svg files with text even if the text is converted to path first

jonnolen, 2015-06-01:
Built an svg with inkscape, choose generic sans-serif font.

path a:
save file as plain svg.
load in mdraw
no image shows

path b:
select text
convert object to path
save as plain svg
load in mdraw
path is garbled.

both files load correctly in chrome, firefox and safari.

Makeblock-official, 2015-06-02:
The svg router is a home brew one, maybe not support all symbols in w3c standard. Would you please upload your svg file and i will look into it.

jonnolen, 2015-06-03:
Here is a sample file:
https://drive.google.com/file/d/0B0Ze0R5lw0KDY0lBZ19RVm9nRUE/view?usp=sharing

This is what it looks like in firefox:
[image]

this is what mdraw thinks it looks like:
[image]

robbo1, 2015-06-04:
Looking at the svg file, mDraw does not know what to do with the "q" state in the path "d" 

The letters without the "q" state rendered correctly

jonnolen, 2015-06-04:
i'm going to attempt to dig into this and make it work.

closed 2015-06-15 by jonnolen