Makeblock-official/mDrawBot #22: Laser Problem

Buitani, 2015-06-13:
I tried the laser on mScara. I would like to cut paper. But it seems like the laser is to weak.
My settings:
Speed: 1%
Power: 254
Delay: 1

What can I do, to cut paper with the laser?

Makeblock-official, 2015-06-15:
maybe you can add delay time, the item delay means additional halt time in ms for each small segment.

Buitani, 2015-07-11:
I tried that too (adding delay time) But it seems sometimes mscara moves with higer delay time even faster. Still can not cut paper...

meeuw, 2015-09-05:
I don't own the drawbot kit but my experience with the xybot is that the setup isn't directly used after changing, I needed to reset the controller by disconnecting the usb and power cable. The "insane" speed might be caused by invalid dipswitch settings for the motor control.

Buitani, 2015-09-05:
I tried your tip. I setup everything in mDraw. High Power and Delay numbers. Then disconnected and reconnected power and usb cable. Result is the same. I cant see a difference on the paper. Laser is always working with the same intensity no matter what numbers I set for Delay and Power.

tur0kk, 2015-12-01:
The problem is that the micro controller firmware (at least the xy plotter) parses the delay time to integer microseconds internally. That means with a value greater than 32ms you have an overflow (jump to -32ms) and the plotter starts to move faster again.

In summary:
you effectively have a delay time from 0ms to 32ms.