Makeblock-official/mDrawBot #31: X and Y reversed for xybot

meeuw, 2015-07-05:
Please have a look at this image:
https://raw.githubusercontent.com/Makeblock-official/mDrawBot/master/mDrawGui/images/xy_setup.png

From this image Y > X

and this code:
https://github.com/Makeblock-official/mDrawBot/blob/master/mDrawGui/XYRobot.py#L99

As you can see:
self.width = 380
self.height = 310

Height < Width

I think the above code is right and the image (and firmware) are now wrong.

I'll try to write a patch this week.

MikeHg, 2015-07-13:
It's not only that xy are changed. I drew a picture with a laser and it seems that sides are also mirror-inverted. It might be that this is only a subsequent error of the bug meeuw reported.

meeuw, 2015-07-30:
@MikeHg could you please test, obviously it works for me :-)

closed 2015-07-30 by meeuw