Makeblock-official/mDrawBot #33: Limit switches not recognized for XY plotter

MikeHg, 2015-07-13:
I did the following. 
1. Go home (i.e. the [0,0] position)
2. Enter values for a certain position and go there, like for instance [100,100]
3. Press "Go home" again.

It might be that it has something to do with x and y swapped as reported by meeuw. But after step 3 the slides will go to one side and the stepper motor continues to make noises without stopping. Only disconneting the plotter and restarting mdraw helps.

Makeblock-official, 2015-07-14:
have you try the latest xybot firmware go with the distribution package or build from the source code, I made some test just now and looks fine.

MikeHg, 2015-07-14:
Yes I used the official distribution (version 1_1.zip) together with its firmware. I also tried the github-master but got the same result.

I didn't use it with the pen but rather with the laser, if that makes a difference.

meeuw, 2015-07-30:
@MikeHg could you try to setup the XY bot motors and set Stepper B in Anti Clockwise mode?
Check the image to see the "home" position and manually hit the limit switches and swap some cables.