Makeblock-official/mDrawBot #35: contribute

schmurfy, 2015-07-17:
is there any information somewhere on how to run it from the sources and build the final executable from it ? (On Mac OS X in my case)

Makeblock-official, 2015-07-17:
I am using mac port to setup my develop environment. here is some tips:
1. install python34, py34-pyqt5 and pyserial by mac port
2. set default python to python34 by port select
3. python mDraw.py should work

the packaged mac distribution by py2app could be more than 100MB, if you have any idea how to reduce the size please let me know, thank you~

schmurfy, 2015-07-20:
thanks, I will try that.

schmurfy, 2015-07-20:
I use Homebrew and, based on your instructions, this worked for me:

```
$ brew install python3
$ brew install pyqt5
$ pip3 install pyserial
$ python3 mDraw.py
```

Can we add these to the README ?

closed 2022-03-03 by schmurfy