Makeblock-official/mDrawBot #40: Command line automation

robertshearing, 2015-09-25:
I'd like to create and automated process where the m

tur0kk, 2015-12-21:
I don't think it is possible to start mDraw with parameters via command line. What you can do is to send gcode commands to the mScara directly via a serial connection. 

```
import serial
import time
# Open grbl serial port
s = serial.Serial('INSERT_YOUR_PORT_HERE',115200)
# Open g-code file
f = open('out.gcode','r');
# Wake up firmware
s.write("\r\n\r\n")
time.sleep(2)   # Wait for firmware to initialize 
s.flushInput()  # Flush startup text in serial input
# Stream g-code to firmware
for line in f:
    l = line.strip() # Strip all EOL characters for consistency
    print 'Sending: ' + l,
    s.write(l + '\n') # Send g-code block to firmware
    s.flushInput() 
    grbl_out = s.readline() # Wait for firmware response with carriage return
    print ' : ' + grbl_out.strip()
# Wait here until firmware is finished to close serial port and file.
raw_input("  Press <Enter> to exit and disable firmware.") 
# Close file and serial port
f.close()
s.close()   
```

Then the only thing you have to do is to translate your image to gcode accepted by mScara.

luispinedaaguirre, 2016-04-17:
@robertshearing were you able to automate the process? if so, can you share the code you use or give me advice on how to do it. I am working on a project that would need automation. Thank you in advance.

luispinedaaguirre, 2016-04-17:
@tur0kk Do you have a C or  C++ code instead of Python that has the same function? Thank you in advance

tur0kk, 2016-04-20:
No sorry, I only have the python code for this basic sending functionality. 

But I wrote a plugin for an open source laser cutter software called [VisiCut](http://visicut.org). It can read in vector files, generate gcode, and send it to a laser cutter. With this plugin it is possible to use [VisiCut](http://visicut.org) with the MakeBlock laser cutter. Find an explanation on my [blog](http://svens-world.de/plotter).