Makeblock-official/mDrawBot #43: mDraw.exe (v1.2) crash with I connect in port COM

tipex20, 2016-03-13:
I've just test the new windows version 1.2 but when I connect in the correct port COM : 
-it write COM3 open success
-the mDraw.exe application kill himself.
-no log :(

How can I do ?
I hope it will work

bigeyex, 2016-03-14:
Hi, 

Would you please upload the mDraw.log file in the exe folder? If there is one.
Please try to open other COM ports and see whether it stops crushing.
I uploaded the old v1.1 version in the distribution folder. Please use it if it's still not working. Thanks.

tipex20, 2016-03-14:
The version 1.1 seems to work with this log

Traceback (most recent call last):
  File "mDraw.py", line 378, in parseRobotSig
  File "D:\mDraw_python\mDraw\ScaraRobot.py", line 350, in parseEcho
IndexError: list index out of range

But the version 1.2 is bugging and no log are in the log file (has if it bug before it can write anything inside)

louis-ev, 2016-05-01:
Hi,
same problem on OSX 10.11.4 with 1.2.1 and 1.2. After selecting the right port, clicking connect makes the app crash. see log here : http://pastebin.com/raw/CU4XaeBF
Version 1.1 works though, fortunately.

timpulver, 2016-05-02:
Yes, I’m also still using v1.1.