Makeblock-official/mDrawBot #45: mDraw crashes when trying to open SVG file

timpulver, 2016-03-13:
mDraw: 1.2
Mac OS X 10.11.3

This happens with most of the SVGs I tried to open (SVG 1.0 as well as 1.1).

Here is an example file:

[Dropbox](https://www.dropbox.com/s/v7q322z5aa7bcr4/line_2016-03-13_21-40-18_all-01.svg?dl=1)

bigeyex, 2016-03-14:
Hi,

This is a confirmed bug. I've uploaded the old V1.1 version in the distribution folder. Please use that version before this bug is fixed.

timpulver, 2016-03-14:
Thanks for the quick response!
I had similar problems with the old version – mDraw didn’t crash, but when I tried to open most SVG-files (like the one I posted above) nothing happens. Seems like mDraw has problems with a lot of SVG-files.
It happened with SVG 1.0 and 1.1, exported from Adobe Illustrator CC.

bigeyex, 2016-03-23:
Hi, 

I've updated version 1.2 in the "distribution" folder. Please have a try.

karstenbondy, 2016-03-26:
1.2 still crashes when I try to open a .svg file. (Mac OS X 10.10.5) 1.1 Still won't load anything. 1.21 also crashes.