Makeblock-official/mDrawBot #46: MDraw crashes when I try to open an .svg file - here is the log

alzakex, 2016-04-05:
Process:               mDraw [5231]
Path:                  /Users/USER/Downloads/mDraw V1.2 201602291936.app/Contents/MacOS/mDraw
Identifier:            org.pythonmac.unspecified.mDraw
Version:               0.0.0 (0.0.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           mDraw [5231]
User ID:               501

Date/Time:             2016-04-05 11:31:29.813 -0700
OS Version:            Mac OS X 10.10.5 (14F27)
Report Version:        11
Anonymous UUID:        6296A409-7268-28BB-E717-6C648D16B2C6

Sleep/Wake UUID:       CE08627E-8CF9-4B29-A916-1862DF029CE2

Time Awake Since Boot: 190000 seconds
Time Since Wake:       1200 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff8a363286 __pthread_kill + 10
1   libsystem_c.dylib             	0x00007fff8fc819b3 abort + 129
2   QtCore                        	0x00000001087783e9 0x108757000 + 136169
3   QtCore                        	0x0000000108779991 QMessageLogger::fatal(char const*, ...) const + 161
4   QtCore.so                     	0x0000000108e21c18 pyqt5_err_print() + 904
5   QtCore.so                     	0x0000000108e272ec PyQtSlotProxy::unislot(void**) + 124
6   QtCore.so                     	0x0000000108e27255 PyQtSlotProxy::qt_metacall(QMetaObject::Call, int, void**) + 85
7   QtCore                        	0x00000001089c596f QMetaObject::activate(QObject*, int, int, void**) + 1359
8   QtWidgets                     	0x00000001097ee0a0 QAbstractButton::clicked(bool) + 64
9   QtWidgets                     	0x000000010957c9ef 0x10947e000 + 1042927
10  QtWidgets                     	0x000000010957c857 0x10947e000 + 1042519
11  QtWidgets                     	0x000000010957d870 QAbstractButton::mouseReleaseEvent(QMouseEvent*) + 272
12  QtWidgets.so                  	0x00000001090eab7e sipQPushButton::mouseReleaseEvent(QMouseEvent*) + 126
13  QtWidgets                     	0x00000001094e31fe QWidget::event(QEvent*) + 1486
14  QtWidgets                     	0x000000010957d5d3 QAbstractButton::event(QEvent*) + 179
15  QtWidgets                     	0x000000010960e678 QPushButton::event(QEvent*) + 88
16  QtWidgets.so                  	0x00000001090eb0b4 sipQPushButton::event(QEvent*) + 116
17  QtWidgets                     	0x00000001094aaffc QApplicationPrivate::notify_helper(QObject*, QEvent*) + 300
18  QtWidgets                     	0x00000001094ae447 QApplication::notify(QObject*, QEvent*) + 8631
19  QtWidgets.so                  	0x0000000108fa49ff sipQApplication::notify(QObject*, QEvent*) + 127
20  QtCore                        	0x0000000108991932 QCoreApplication::notifyInternal(QObject*, QEvent*) + 114
21  QtWidgets                     	0x00000001094ab8e8 QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) + 952
22  QtWidgets                     	0x000000010950156c 0x10947e000 + 537964
23  QtWidgets                     	0x000000010950084f 0x10947e000 + 534607
24  QtWidgets                     	0x00000001094aaffc QApplicationPrivate::notify_helper(QObject*, QEvent*) + 300
25  QtWidgets                     	0x00000001094adabb QApplication::notify(QObject*, QEvent*) + 6187
26  QtWidgets.so                  	0x0000000108fa49ff sipQApplication::notify(QObject*, QEvent*) + 127
27  QtCore                        	0x0000000108991932 QCoreApplication::notifyInternal(QObject*, QEvent*) + 114
28  QtGui                         	0x00000001082ab5fc QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 2140
29  QtGui                         	0x00000001082aa565 QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) + 117
30  QtGui                         	0x00000001082991cb QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 315
31  libqcocoa.dylib               	0x0000000109cbef0d QCocoaEventDispatcherPrivate::processPostedEvents() + 317
32  libqcocoa.dylib               	0x0000000109cbf8a8 QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void*) + 40
33  com.apple.CoreFoundation      	0x00007fff9183ca01 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
34  com.apple.CoreFoundation      	0x00007fff9182eb8d __CFRunLoopDoSources0 + 269
35  com.apple.CoreFoundation      	0x00007fff9182e1bf __CFRunLoopRun + 927
36  com.apple.CoreFoundation      	0x00007fff9182dbd8 CFRunLoopRunSpecific + 296
37  com.apple.HIToolbox           	0x00007fff9339356f RunCurrentEventLoopInMode + 235
38  com.apple.HIToolbox           	0x00007fff933931ee ReceiveNextEventCommon + 179
39  com.apple.HIToolbox           	0x00007fff9339312b _BlockUntilNextEventMatchingListInModeWithFilter + 71
40  com.apple.AppKit              	0x00007fff8c7c88ab _DPSNextEvent + 978
41  com.apple.AppKit              	0x00007fff8c7c7e58 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 346
[... 466 lines cut ...]
__IMAGE                             528K
__LINKEDIT                         76.7M
__TEXT                            183.0M
__UNICODE                           552K
mapped file                        67.8M
shared memory                         4K
===========                      =======
TOTAL                             700.0M
TOTAL, minus reserved VM space    699.7M
 

Model: MacBookPro11,3, BootROM MBP112.0138.B15, 4 processors, Intel Core i7, 2.5 GHz, 16 GB, SMC 2.19f12
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.15.166.24.3)
Bluetooth: Version 4.3.6f3 16238, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500.28 GB
USB Device: Internal Memory Card Reader
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1

timpulver, 2016-04-13:
Crashes here, too. Sometimes when I click on the dropdown to select «XY Plotter», sometimes when I click on «Open» to select an image. Cannot get beyond that.

Using Mac OS X 10.10 Yosemite.

timpulver, 2016-04-13:
Seems like the crash has something to do with the serial connection. Shortly after I click «Connect» it crashes… :(