Makeblock-official/mDrawBot #47: Mac release "mDraw V1.2.1 201603222109" only lets you choose mScara

algal, 2016-04-09:
On OS X 10.11.4, you can run the release, but you cannot use the drop-down menu to change to any mode besides "mScara".

algal, 2016-04-11:
Follow-up on this. I've realized that what is actually happening is that certain mDrawBot UI elements are frozen for 2-3 minutes after it is launched. This includes the dropdown picker for robot configuration (to change from mScara to any other configuration) and the dropdown picker for choosing the communication port (to change to `/dev/tty/wchusbserial1410`).

So if you wait long enough, then you _can_ choose configurations besides mScara. I am running on a MacBook Air (Mid 2012) on OS X 10.11.4

I am still not able to successfully update firmware, but that is another story.

InevitableCraftsLab, 2016-10-24:
Hi, did you ever found out why you can only choose the scara?

I can choose it, but it never stays when power on/off

Also the scara doesn't remember the settings, and none of the versions of mdraw works on 10.11, except the first one.