Makeblock-official/mDrawBot #52: OSX version not working except scara

InevitableCraftsLab, 2016-10-25:
Hi is there any plan of releasing a running version of mDraw for OSX?

So far your current software for osx is a desaster.
Unsigned drivers, 
no usb connection, 
not able to upload the firmware, 
not able to use any other model then mscar,
not able to save preferences

the oldest version (1.0) works somehow as long as you dont use any other then the mscara model
the 1.2 version does nothing
the 1.2.1 version hangs up while starting.

is there any more unprofessional way to sell an arduino and two nema17 steppers and charge 200$ ?

i dont think so

and just for the record: this is not because the product is for tinkerers and diy-people, because even they want a product that does AT LEAST whats in the products description. 

tinkering, is the process you start after building and running it.

bigeyex, 2016-10-25:
what version of MacOS are you using? No issue so far even in the latest MacOS Sierra.
[image]

InevitableCraftsLab, 2016-10-25:
I am running el capitan, and your screenshot shows the mscara, which is working fine.
Everything else then the mscara doesnt work for me, no mdraw version can update the firmware.
I assume that the firmware is still in scara mode, even when the mDraw shows its in spider mode.

I attached a screenshot that shows what the spider does when it should draw the concentric circles
[image]

InevitableCraftsLab, 2016-10-25:
i just upgraded to sierre, still the same issues.
I know its probably not you personal fault, but i will never ever buy a product from your company, this is the combination of a ripoff and a joke...
[image]