Makeblock-official/mDrawBot #54: Makeblock XY Plotter enlarged.

cutplotter, 2017-03-28:
I am a newby in Makeblock and mDraw.
I have bought a XY Plotter v.2.
I like much more mDraw intead of Benbox.
The problem with Benbox are many:
             - Don`t allow to resize placed objects (mDraw yes).
             - Don`t allow to place objects with exact coordinates (mDraw yes, using the coordinates in upper rght corner of the program to place the head and then the object).
             - Don`t allow to place the plotter picking with the mouse in any place of the working area (mDraw allow this, and is very useful to test if the design fits into the paper).
               Benbox only allow to move the plotter in “manual” mode with the arrows in the upper left corner of the program, but you never know in what “coordinates (x-y)” the plotter are... (mDraw yes).
            - You can`t reverse motor function to place the origin where you want (mDraw yes).
            ....
 
So, I need to enlarge the working area of XY Plotter to 1000x700 mm.
To get this I need to add 2 functions into mDraw:
 
1.- Positioning of an imported “svg” with coordinates (by the keyboard) not with the mouse that is manual and inexact.
 
2.- The maximum area wich I can see the borders of the working area in mDraw is 740x690 mm. If I increase that area, I can`t see the borders of the working area in mDraw window. So I need a “zoom” option (to zoom in and out).
 
 
Anybody knows if somebody create some of this functions for mDraw before?
If not, can anybody add this 2 options for me?
Or can anybody teach me how to do it?
  
Thanks.
Regards,

joanielemercier, 2017-03-28:
Hi,
I have exactly the same needs, and mdraw was never finished by the developers. 
I'd be really interested to know if what you listed is possible, and happy to help.

bigeyex, 2017-04-07:
Makeblock is making a better software that will replace the XY plotting functions of mDrawBot.

At current stage, mDraw will only support mDrawBot in the default form. However, as an open source project, help from all sides are welcomed.

joanielemercier, 2017-04-07:
@bigeyex  Thanks for the update.
Would it be possible to have early access to the new software ?

I just made a 830mm x 390mm, and with the problems listed above I'm stuck and I can't use the plotter. :(
I wish I could recompile mdraw myself, but I'd need some help as well.

joanielemercier, 2017-04-08:
[image]

cutplotter, 2017-04-08:
Ok joanielemercier, 
You enlarge only one of the axes around +500 mm, more or less.
But you can`t work with it because the limitations of mDraw, right?
The movement in 830 goes soft?
Don`t you have the 8 mm axe little cruved for this distance?

joanielemercier, 2017-04-08:
Hello @cutplotter 

Hardware works great, no bending/curve at all, movement is super smooth, and resolution is great (I have maybe 0.1mm curve on the of the angles in the center, but I'm working on it !
I'm now working on doubling the Y as well to achieve 80cm x 80cm bedsize.

The only issue is with mdraw (svg scale problem, Y axis motor bug (the pen sometimes goes -Y instead of Y, and with limiter switch not  working there's a high risk of breaking everything, and window beeing too small.

I wish I knew how to recompile mdraw and fix all this, it's so frustrating.

Here are some details of the center.
[image]