Makeblock-official/mDrawBot #59: mDraw Mac, XY Plotter: `Abort trap: 6` when opening SVG file

moeffju, 2018-01-11:
Using `mDraw V1.2.2 201607201019.app` with the XY Plotter, some SVG files cause mDraw to crash when opening the file. All I get on the console is `Abort trap: 6`.

System logs only give me these two lines:

```
Jan 11 12:36:54 Sabre mDraw[87330]: assertion failed: 16G1114: libxpc.dylib + 74307 [BF896DF0-D8E9-31A8-A4B3-01120BFEEE52]: 0x87
Jan 11 12:36:55 Sabre mDraw[87330]: notify name "user.uid.501.BRNotificationServerAvailabilityChanges" has been registered 20 times - this may be a leak
```

How can I get more debugging info or fix this error?

ploenne, 2019-03-04:
fix is here: https://github.com/Makeblock-official/mDrawBot/pull/60
but it seems nobody cares anymore
you can try to convert all Uppercase 'Z' to lowercase 'z' in the svg file, using a simple texteditor.

moeffju, 2019-05-17:
cc @bigeyex or anyone still active here?